AdamParlane/aut-swarm-robotics #8: Why does RXRDY for TWI0 need to be cleared at setup time?

wittsend, 2017-07-17:
Was browsing the code and noticed that there is a line that loads a "dummy" value into the receive hold register for TWI0 after setup. What is this for?
`uint8_t dummy = REG_TWI0_RHR;			//Ensure RXRDY flag is cleared`
this is in `twi0Init()`

Just curious...

AdamParlane, 2017-07-19:
Not 100% sure, thats more of an Esmond / Jake question (not sure which one wrote it), but I just presumed you had to clear it to get going, im guessing though

wittsend, 2017-07-19:
Haven't seen this in any example code. Will ask Esmond when I see him.

EsMather, 2017-07-21:
Hey,
It was just used when we were troubleshooting the TWI. You could delete that line. I'm sure it will still work without it.

wittsend, 2017-07-21:
OK thanks Es

closed 2017-07-21 by wittsend