AdamParlane/aut-swarm-robotics #9: Move hardware setup routines

wittsend, 2017-07-17:
Move hardware setup routines out of main.c and into their respective modules.
Also, where to put generic setup functions like master clock and PIO setup (granted most PIO will be handled by specific drivers, but there are specific things like the LEDs).
Another thing is, is this the best way to do this? ie, should the hardware setup all be done it is own module with setup functions for each bit of hardware. I think it would make the hardware setup functions easier to find. What do you guys think about it?

AdamParlane, 2017-07-19:
For the most part they are already moved out, exceptions are some of the timer stuff which I think is for your IMU that I left for you to move so I didnt break it, and the LEDs, PIO and master clock I was thinking would be okay just there in a master setup function as it currently is. However this function could move to a different .c file, although I'm not sure how useful it could be.

I prefer keeping the setup of each hardware in the respective hardware .c files.
Im not sure what the industry standards are but my reasons are:
(1)Portability, In theory I could take my opt_interface.c and  .h put it on a SAM4S chip and it will just work. If the setup file is somewhere different then i'd be missing the key part to run that hardware, sure I could rewrite it or copy it, but whats the point. 
(2) I see what you are saying about keeping them together to be easy to find, but in general at least from what I can imagine if you are debugging the mouse (for example) then you would be looking at the mouse functions and the setup so you want them together. There wouldnt be may cases where you just want to see how each thing is setup up without the context of how it is being used.

wittsend, 2017-07-19:
I think having all hardware setup elsewhere just helps make it more modularised. I think that the setup function in main.c should only be calling other setup functions (for each piece of hardware) rather than having specific setup code in it.

As for keeping hardware setup where they are, I think you make a good point about portability. I guess my concern lay mainly with the TWI2 setup being with the IMU, where in future it might not necessarily be just for the IMU.. (I may be wrong but off the top of my head, I think external TWI is on TWI2?)

wittsend, 2017-07-20:
I know what we can do. We'll just list where each hardware setup routine can be found in the main.c header. That way they'll be easy to find.

wittsend, 2017-07-25:
Done

closed 2017-07-25 by wittsend