AdamParlane/aut-swarm-robotics #21: Two versions of code

wittsend, 2017-07-21:
Esmond says that we'll have 4 robots up and running very soon, and that we will need two versions of code for the old and new bots due to hardware revisions between the two. I propose that instead of two whole versions of code, we just have two separate project files that share source code. Each project file will have a compiler directive that states what robot version it is for and the code will have compiler 'if' statements that determines which hardware setup to configure the uC for. That way we are still only working out of one set of source code files instead of two.

AdamParlane, 2017-07-21:
Absolutely agree, as soon as I read your first line I started thinking along the sdame way. I wonder if we even need 2 project files or just a line where you select which robot type you are using. Or better yet its possible to do that automatically, but I only know how to that with different chips wonder if there is some ID we could hardcode in to make it automatically choose, ill start looking into it

wittsend, 2017-07-21:
I don't think its any trouble to have separate project files for different robot revisions. Whatever robot you have in front of you for the day, you just make sure you load up the project for that revision. Having it auto-detect some kind of identifier would mean that both versions of code would still have to be loaded onto the microcontroller because the correct code would be selected at runtime rather than compile time.

EsMather, 2017-07-22:
Hey I just put the new Pinouts up on onenote under 'Robot Hardware > SAM4N Pin Changes'.
It looks like most of the new pin changes are for the new camera buffer, so you probably only need to change a handful of your current #defines relating to things like the motors and line followers. I'll also get the latest schematic off Jake and put it on that page as some of the pin labels are extremely vague, unless you are specifically looking for it.

wittsend, 2017-07-23:
There are now two project files within the solution. One for the old robots (V1) and one for the new (V2). They both point to the same source code files. When you open the solution (now called swarm_robot.atsln) you will see both projects on the right hand side in the solution explorer. You can build each project independently in the build menu, but I'm not sure how you determine which one gets loaded onto the robot when programming yet. The next thing to do is add the compiler statements to the code and the specific compiler directives to the project files.

wittsend, 2017-07-25:
Added a page on one note under software that details how to build and execute code for the robot that you are using.

closed 2017-07-25 by wittsend