AdamParlane/aut-swarm-robotics #66: twi read functions need to detect 'not acknowledge' so that program doesn't freeze when I2C slave doesn't reply

wittsend, 2017-08-02:
Functions should return an error code of some sort to indicate that there was a problem.

AdamParlane, 2017-08-02:
I like this, I was thinking last night that when we do a whiteboard session of all the modules and their interactions we should add some error detection and reporting functionality including exactly this (probably our most common run time error at this stage). Then the bot can tell the PC it had a function that failed etc and it can be graveyarded if happens again (or once for a critical error). I think we should just whiteboard all this next tuesday and before then we should both have a little think about what we need to make it modular and all our respective modules work well. If I have time I might try start this sooner

wittsend, 2017-08-10:
TWI functions will no longer freeze up on error. The time out values for the TWI can be set by constants at the top of twimux_interface.h

closed 2017-08-10 by wittsend