AdamParlane/aut-swarm-robotics #74: Delay Function for TWI Bitbang

EsMather, 2017-08-04:
Hey I've been writing some bit bang code for the camera. But it keeps locking up in the delay function.
I was wondering if the delay function requires comms with the IMU? and is there a simple way for me to get the delay in 'uS' instead of mS?

wittsend, 2017-08-05:
The short answer is no. The delay function uses timer0 which is initialised
with the IMU at the moment. The problem is, that timer0 is used for the
clock signal to the camera, so it won't work with the delay function.
(Camera clock is 12MHz whereas I had timer0 running at 1kHz). If you are
already using timer0 for the cameras 12MHz clock, then you just need to add
code that increments systemTimestamp once every ms. You could try to make
another counter that increments once every us, but I imagine it will
overflow very quickly.

<https://github.com/AdamParlane/aut-swarm-robotics/issues/74>, or mute the
thread

AdamParlane, 2017-08-05:
Just checking: this means that I need to change my use of it now too because system time stamp is incremented every 83ns instead of every 1ms?

wittsend, 2017-08-05:
System timestamp needs to increment every 1ms because it is used by
delay_ms and get_ms which are both needed for the IMUs DMP driver. So I
guess you would need to rewrite the interrupt for timer 0 so that system
timestamp still incremented every one msec and then add a new counter that
incremented every 1us for bit banging.

AdamParlane, 2017-08-06:
Resolved with Merge #80, timer is now its own module. Interrupt now handles us and ms. Delay_ms and get_ms are untouched and intact. Camera frequency is still as is at 12.5MHz and delay_us is available to bit bang.

closed 2017-08-06 by AdamParlane