AdamParlane/aut-swarm-robotics #75: Overall Project Needs to be made more modular

AdamParlane, 2017-08-06:
Adam and Matt to Whiteboard and resolve

wittsend, 2017-08-09:
I think that we are going to have to do away with having all the includes in robot_defines (other than ones that all functions definitely need, like sam.h) because we actually don't know what the connections are between the modules. If we just have the includes at the top of each file for only the stuff they need, then we can quickly see what the connections between the modules are. Plus we can add commenting to explain why certain connections and includes have been made on a per module basis.
An example this morning was that I am having difficulty now sharing the robotPosition structure with other modules without getting warnings or errors due to where it is located relative to all the includes in robot_defines. Ended up having to place the type definition for Position above the includes, which isn't really best practice.

AdamParlane, 2017-08-16:
Adding this here so we can work to it as we go

[image]

wittsend, 2017-08-27:
Bump

Wouldn't mind starting on giving all the modules their own includes shortly. What do you reckon?

wittsend, 2017-09-15:
Issue resolved:

robot_setup.h only contains the following includes which are used by all modules in the program:

```
#include "Interfaces/spi.h"		//Fixes SPI issue
#include "sam.h"			//Microcontroller specific defines
#include <stdint.h>			//Gives standard integer type definitions (ie uint8_t)
#include <stdbool.h>			//Gives boolean variable types
```

All other module header files no longer contain #include directives, as this is bad practice. #includes reside solely in .c files. Each .c file only includes headers to modules that they need. robot_setup.h should be included FIRST in every module so that they all have access to the RobotGlobalStructure data structure. The include for a modules own header should be last, in case it needs access to the references in the preceding includes. (At very least it should be after robot_setup.h)

closed 2017-09-15 by wittsend