AdamParlane/aut-swarm-robotics #90: Building the timeout function into your code

wittsend, 2017-08-10:
There is a new function in robot_defines.c called waitForFlag(). It should be used where ever you would normally place a while loop to wait for a register flag. Basically it will wait for your flag only as long as you specify. If it doesn't see the flag within the time limit, then waitForFlag() returns a 1. The syntax to use this function is as follows. Its a little tricky because we are passing a register address to it which can change dynamically:

```
//wait for start and data to be shifted out of holding register
if(waitForFlag((uint32_t*)&REG_TWI0_SR, TWI_SR_TXRDY, TWI_TXRDY_TIMEOUT))
	return 1;
```

The above example is taken from one of the TWI driver functions. The first parameter is the register you want to wait for. You MUST have that type casting in front of the register name for this to work. The second parameter is the bit mask for the register. The register is bitwise ANDed against this to only show the bit(s) you want to wait for. The final parameter is the time to wait in milliseconds. It should be placed in an if statement so that you can tell the program what to do if the function times out. I suggest making it exit the current function with a non zero value as above. Happy error trapping :-)

AdamParlane, 2017-08-10:
Good stuff, however the way I see it, this shouldnt be in issues, but prehaps in the readme. Or contributing

wittsend, 2017-08-10:
Yeah it's just here because I know this is read more regularly than readme.