AdamParlane/aut-swarm-robotics #133: Global Data Structure

wittsend, 2017-09-14:
See the 'Docking' branch. Haven't merged because its untested on robot + has a couple of warnings
See below for the layout of the global structure:

/*Global Data Structure Initialisation
Current Layout Tree:
--------------------
+sys
	+flags
		+xbeeNewData
		+imuCheckFifo
		+obaMoving
		+obaEnabled
	+states
		+mainf
		+mainfPrev
		+docking
		+chargeCycle
		+followLine
		+scanBrightest
		+moveHeadingDistance
	+pos
		+Optical
			+dx
			+dy
			+heading
			+speed
		+IMU
			+qw
			+qx
			+qy
			+qz
			+accelX
			+accelY
			+gyroX
			+gyroY
			+gyroZ
			+pitch
			+roll
			+yaw
			+yawOffset
			+timeStamp		//Might move these up to pos
			+deltaTime		//
		+x
		+y
		+heading
		+targetHeading
		+targetSpeed
*/

wittsend, 2017-09-15:
In the future, I want to see the charging state and battery state in here, and colour and prox data stored in here. I also want there to be an enable flag for different peripherals so that we can turn off polling them when we don't need them (so only poll the FC chip when we are looking for the dock and charging, and not any other time for example.). That way, Instead of having to call functions to read the FC chip from docking_functions, we simply enable the FC_chip enable flag in the RobotGlobalStructure, and the next time around the main loop, it is polled and the new data loaded into the structure. Then when finished, we disable polling the FC chip. I think it will neaten up code a lot.

NZBanana, 2017-09-18:
Is there anything dependent on using c over c++? If not it might be nicer to use c++ and make use of object orientated programming? With our timeline this is unlike to help this year but it could be listed as an improvement

wittsend, 2017-09-19:
Good point, but I think we'll leave that for next year :)

wittsend, 2017-10-03:
Totally implemented now

closed 2017-10-03 by wittsend